gazcbm/Marlin-2.0.x-SKR-Mini-E3-v1.2 #1: BLtouch Version and Wiring

nickwebcouk, 2019-10-12:
Can you confirm which version of BLTouch this firmware is for and what port its plugged into on the board? I have the 3.1 bought from reality and it has different coloured wiring - and plugged into the servo port but its not deploying using the menus. The auto test at startup is fine so I figure its another setting to be changed somewhere. Thanks.

gazcbm, 2019-10-13:
My BLtouch was a version 2, and initially I had the same issues and it was just a case of reading the docs.  In my case I found that the official cabling was Brown Gnd, Red 5v, Yello Servo.  For the SKR board you can confirm the pins here https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/hardware/BTT%20SKR%20MINI%20E3%20V1.2/BTT%20SKR%20MINI%20E3%20V1.2PIN.pdf . 

According to the BL touch 3.1 manual the wiring is the same!??.. https://docs.wixstatic.com/ugd/f5a1c8_d40d077cf5c24918bd25b6524f649f11.pdf

gazcbm, 2019-10-13:
If the BLtouch is lighting up, you have at least the 5v wire in the right place.

nickwebcouk, 2019-10-14:
Yeah. After all that it was the two outer wires that were reversed. For your readme file - my Reality ordered Official BLTouch 3.1 has the following three wires into the servo port

`YELLOW BLUE RED`
`--GND--+5V--PA1`

Thanks for your help!

closed 2019-10-14 by nickwebcouk