gazcbm/Marlin-2.0.x-SKR-Mini-E3-v1.2 #4: Board will not boot when bltouch is connected to servo ports

Trevor159, 2019-10-17:
I understand this is not your issue at all but I thought you might understand this a little more than me. 

https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/43

This is the issue I made on the official github repo but my v1.2 board only shows backlight when the bltouch is connected to the servo ports. Maybe I am stupid but this seems like a hardware issue and not a software issue from my observations but I might just be dumb.

gazcbm, 2019-10-17:
What colour is your middle pin on the dupont connector?

Trevor159, 2019-10-17:
Solved: the wiring diagram for the creality bltouch in your readme had the red and blue wires flipped. I had a creality v3.1 fwiw

closed 2019-10-17 by Trevor159

gazcbm, 2019-10-18:
Wow it seems like creality have many different flavours of wiring.  Thanks for the headsup, I will add your info to the readme