gazcbm/Marlin-2.0.x-SKR-Mini-E3-v1.2 #6: A lot of probing

impy101, 2019-10-24:
First and foremost: AMAZING work. It works surprisingly well. The only reason why I used the word surprising is because I'm coming from BTT's shoddy firmware. 

My only issue with this firmware is that the BL touch probes 49 times before it starts the print. I checked the configuration.h file and I honestly can't seem to figure out how to reduce it. I come from Th3D's firmware with only 9 probes, you see..

Some help would be much appreciated.

gazcbm, 2019-10-25:
Hey Impy101, checkout the new bltouch-improvements branch, its a 3x3

gazcbm, 2019-10-25:
Will eventually merge into master

closed 2019-10-25 by gazcbm

impy101, 2019-10-25:
You, sir, are awesome! I will bookmark you for future updates etc. Hopefully you will add a version with filament runout protection alongside ABL since I just received my filament runout protection kits. :)