gazcbm/Marlin-2.0.x-SKR-Mini-E3-v1.2 #7: Marlin 2 and BLTouch probing grid

meltonpieman, 2019-10-25:
Thanks for this great branch for the Mini E3 and the BLTouch. Just a question (I am new to Marlin 2 -forced here by the Mini E3). 
In Marlin 1.9 I selected ABL as Bilinear set the X&Y probe offsets from the nozzle and the Minimum Probe Edge for the bed. Then to compensate for the Probe offset I would define something like:
RIGHT_PROBE_BED_POSITION (X_BED_SIZE - (MIN_PROBE_EDGE +[the offset+2mm])) etc

I notice that you have not uncommented the 4 probe bed postion statements in your confuguation.h file. lines 1206-1210

Am I missing something elsewhere that does the min and max postioning? If so, sorry but can you point it out to me?

meltonpieman, 2019-10-27:
I have returned to your original version and tried to recompile it and suddnely I get an error that I do not understand:
xPack OpenOCD, 64-bit Open On-Chip Debugger 0.10.0+dev (2019-07-17-11:28)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1

hla_swd
Error: open failed
in procedure 'program'
** OpenOCD init failed **
shutdown command invoked

*** [upload] Error 1

Have you got any suggestion?

Slevin7, 2019-12-17:
Did you figured out what the problem was?
Cause i have the same error

shawnb2, 2019-12-19:
No. Never did. Went back to the Melzi board and loaded in the TH3D firmware. It’s been working flawlessly. Not gonna change what’s not broken at this point. Maybe once my Prusa Mini shows up and have a second printer I’ll start tinkering again.