gazcbm/Marlin-2.0.x-SKR-Mini-E3-v1.2 #8: Changing Probe Offsets then Recompiling

eeynooo, 2019-10-27:
So I need to change the probe offset from the nozzle because my BLTouch will probe the last 3 off the bed, so It says to change them in the config.h which I did, but I've been trying to figure out how to recompile this to get a .BIN file and for the life of me I cannot get it to happen, I have platformIO, and All i did was download the zip, extract it, open it in platformIO and go to the config change the numbers, and tried to compile it using ctrl+alt+B and I get an error. I'm kinda new to this so sorry if it is something super basic, Ive just spent hours on this stupid thing

gazcbm, 2019-10-28:
Hi, I will need a bit more info.  However I expect its one of the following;
1) The path you have stored the source code is probabaly too large or on a "online" drive such as onedrive.. Try something simple like c:\marlin\
2) You will likely need to install git bash

If all else fails, dump your log into pastebin and include the link here, I will try to take a look

rickard2014, 2020-04-06:
I've attempted to change the configuration.h and then recompile, only to adapt the offsets to suit the hero me fan duct, but after flashing the firmware, the printer probes only on the edge when homing, which is weird because the compiled firmware works 100%. Any ideas what's wrong?